jantje/hardware #1: Debugging arduino with Eclipse

kshok, 2015-07-05:
My target is arduino1.6.1 debugging that use eclipse luna.
* Eclipse IDE for C/C++ Developers 64bit
  Version: Luna Service Release 2 (4.4.2)
  Build id: 20150219-0600

* plugin :  Arduino Eclipse Extensions	2.4.0.201507010214 it.baeyens.arduino.feature.feature.group	Jan Baeyens
              C/C++ Development Tools	8.6.0.201502131403	org.eclipse.cdt.feature.group	Eclipse CDT
* Arduino1.6.1
* win7: 64bit

I was follow  http://blog.baeyens.it/#comments23 .
I have errors this step [Right click the project->build configurations->build all.]
Capture 
[image]
Capture Problems
[image]

jantje, 2015-07-06:
please provide the console output.
The compiler should not take MinGW when you selected debug in the boards settings. It could if you take unit test setting.
http://iloapp.baeyens.it/blog/blog?ShowFile&image=1435787452.png

jantje, 2015-07-06:
Going again through the errors I see you have not properly configured the eclipse plugin. Again see the console output. It should be self explaining.

kshok, 2015-07-07:
Yes, here.
[image]

Is thought strangely.
so,
change board.txt file to "C:\Arduino\hardware\arduino\avr\boards.txt
[image]
Have a good day!!

kshok, 2015-07-07:
and Installation Details
[image]

kshok, 2015-07-07:
and console output

12:16:53 **** Incremental Build of configuration Release for project debugDemo ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\Arduino\hardware\jantje\avr_local\cores\arduino" -I"C:\Arduino\hardware\jantje\avr_local\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:88:0,
                 from C:\Arduino\hardware\jantje\avr_local\variants\standard/pins_arduino.h:28,
                 from C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:17,
                 from ../.ino.cpp:6:
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/io.h:612:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from C:\Arduino\hardware\jantje\avr_local\variants\standard/pins_arduino.h:28:0,
                 from C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:17,
                 from ../.ino.cpp:6:
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:109:0: warning: "PROGMEM" redefined [enabled by default]
 #define PROGMEM __ATTR_PROGMEM__
 ^
In file included from ../.ino.cpp:6:0:
C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:14:0: note: this is the location of the previous definition
 #define PROGMEM
 ^
In file included from ../.ino.cpp:6:0:
C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:223:0: warning: "pgm_read_byte" redefined [enabled by default]
 #define pgm_read_byte(a) (*a)
 ^
In file included from C:\Arduino\hardware\jantje\avr_local\variants\standard/pins_arduino.h:28:0,
                 from C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:17,
                 from ../.ino.cpp:6:
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1046:0: note: this is the location of the previous definition
 #define pgm_read_byte(address_short)    pgm_read_byte_near(address_short)
 ^
../.ino.cpp:8:2: error: #error the file: debugDemo.ino is not found in the indexer though it exists on the file system.
 #error the file: debugDemo.ino is not found in the indexer though it exists on the file system.
  ^
../.ino.cpp:9:2: error: #error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
 #error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
  ^
../.ino.cpp:10:2: error: #error please check wether *.ino and *.pde are marked as C++ source code in windows->preferences->C/C++->file types.
 #error please check wether *.ino and *.pde are marked as C++ source code in windows->preferences->C/C++->file types.
  ^
In file included from C:\Arduino\hardware\jantje\avr_local\variants\standard/pins_arduino.h:28:0,
                 from C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:17,
                 from ../.ino.cpp:6:
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1167:26: warning: 'strlen_PF' initialized and declared 'extern' [enabled by default]
 extern size_t strlen_PF (uint_farptr_t src) __ATTR_CONST__; /* program memory can't change */
                          ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1167:26: error: 'uint_farptr_t' was not declared in this scope
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:91:49: warning: attributes after parenthesized initializer ignored [-Wattributes]
 #define __ATTR_CONST__ __attribute__((__const__))
                                                 ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1167:45: note: in expansion of macro '__ATTR_CONST__'
 extern size_t strlen_PF (uint_farptr_t src) __ATTR_CONST__; /* program memory can't change */
                                             ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1168:27: warning: 'strnlen_PF' initialized and declared 'extern' [enabled by default]
 extern size_t strnlen_PF (uint_farptr_t src, size_t len) __ATTR_CONST__; /* program memory can't change */
                           ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1168:27: error: 'uint_farptr_t' was not declared in this scope
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1168:53: error: expected primary-expression before 'len'
 extern size_t strnlen_PF (uint_farptr_t src, size_t len) __ATTR_CONST__; /* program memory can't change */
                                                     ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:91:49: warning: attributes after parenthesized initializer ignored [-Wattributes]
 #define __ATTR_CONST__ __attribute__((__const__))
                                                 ^
C:\Arduino\hardware\jantje\avr_local\variants\standard/avr/pgmspace.h:1168:58: note: in expansion of macro '__ATTR_CONST__'
 extern size_t strnlen_PF (uint_farptr_t src, size_t len) __ATTR_CONST__; /* program memory can't change */
                                                          ^
[... cut ...]
C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:140:58: warning: '__progmem__' attribute directive ignored [-Wattributes]
 extern const uint8_t PROGMEM digital_pin_to_bit_mask_PGM[];
                                                          ^
C:\Arduino\hardware\jantje\avr_local\cores\arduino/Arduino.h:141:55: warning: '__progmem__' attribute directive ignored [-Wattributes]
 extern const uint8_t PROGMEM digital_pin_to_timer_PGM[];
                                                       ^
make: *** [.ino.cpp.o] Error 1

12:16:57 Build Finished (took 3s.323ms)

jantje, 2015-07-07:
this is the config error I referred to above
>../.ino.cpp:8:2: error: #error the file: debugDemo.ino is not found in the indexer though it exists on the file system.
#error the file: debugDemo.ino is not found in the indexer though it exists on the file system.
^
../.ino.cpp:9:2: error: #error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
#error this is probably due to a bad eclipse configuration : ino and pde are not marked as c++ file.
^
../.ino.cpp:10:2: error: #error please check wether *.ino and *.pde are marked as C++ source code in windows->preferences->C/C++->file types.
#error please check wether *.ino and *.pde are marked as C++ source code in

jantje, 2015-07-07:
apperantly somthing is wrong with the variable "compiler.path" and it is simply missing.
The MinGW compiler found on your system seems to introduce problems.
I'm looking for the root cause of the missing "compiler.path" variable.
As a work around you could set the arduino delivered gcc compiler in your path before MinGW

jantje, 2015-07-07:
I have done some investigation:
I started with using the gcc compiler delivered with arduino. However when I started to add functionality like adding input and output to files for the stream objects I had to switch to a "system delivered gcc".
To do so I simply removed the "compiler.path" variable.

This means that on your system in the path there needs to be a full blown gcc compiler that can generate code for your local system. 
My current thinking is that the gcc compiler you currently use is a avr targettted platform.

All the errors related to avr_local\variants\standard/avr/pgmspace.h are self inflicted.
Why? You must have added a pgmspace.h file as there is none in the github repository. 
I know the lack of the file pgmspace.h  causes warnings .... but no errors.
To get rid of the warning make an empty pgmspace.h but do not simply copy a file from "somewhere"

kshok, 2015-07-07:
I do not know whether I understood properly. 
Anyway, I try delete  [C:\MinGW\bin;C:\MinGW\msys\1.0\bin;] in system  environ  path .
result is
[image]

Maybe. I don't understood [missing "compiler.path" variable].
What can I do for next?

jantje, 2015-07-07:
what to do:
1) Install gcc for your os (as you are on windows I don't know what you should use)
2) remove the pgmspace.h you added to the cores folder.

kshok, 2015-07-08:
1) Install gcc for your os (as you are on windows I don't know what you should use)

Yes I did install MinGW for Win7.
Because MinGW a GCC Compiler for Windows 7 (http://www.mingw.org)

2) remove the pgmspace.h you added to the cores folder.
Yes I remove the pgmspace.h.

and result is
[image]
[image]

What can I do for next?

kshok, 2015-07-08:
In addition, I did these and other test, but result is bad.

jantje, 2015-07-08:
This looks far better :-).
Note that for problems the console is the only thing that counts.
But anyway in the problem view there are only 3 errors and 1 warning.
The warning has to do with past actions you did. I would bet on adding the spi library and then deleting the library folder in the project. This should fix itself over time and is harmless.
The make error is harmless and a known issue.
So I think the only problem left is that boolean is declared 2 times. Fix-> comment line 19 of arduino.h

kshok, 2015-07-14:
Thank you for your reply.
Eexecute by your information, other file omissions are repeated.
Do not go forward more, have to look for alternative way if want to finalise business once.
But, I wished to use your useful Debugging Plugin.
Seem to be used well to LINUX, try to take a few minutes and try.
Of course, last should uses in Win7 environment.
System that use now is as follows.

Win7 64bit
MinGW 4.8.1
Eclipse Luna CDT  -> mars
Arduino 1.6.1 -> 1.6.5-r2

Thank you at your information.
This writing was translated to translation machine.

jantje, 2015-08-27:
all the items discussed here are now fixed. tested on windows 8.1

closed 2015-08-27 by jantje