jantje/hardware #2: Debugging arduino with EclipseMars for win7 ,vitualbox 5.0, ubuntu 14.04

kshok, 2015-07-18:
My target is arduino1.6.1 debugging that use eclipseMars for  win7 ,vitualbox 5.0, ubuntu 14.04 .
   • Eclipse IDE for C/C++ Developers mars 64bit
   • plugin : Arduino Eclipse Extensions 2.4
   • Arduino 1.6.1
   • win7: 64bit
   • Oracle VM VirtualBox 5.0
   • Ubuntu 14.04

I was follow http://blog.baeyens.it/#comments23 .

I have success this step
 [Compile the project in the debug configuration (this is not strictly needed but it makes it easier to fill in the following forms)]
[image]

next step  [Now we'll create a debug configuration for our debugDemo project using the "jantje hardware".]
[image]

also next step is fail . [Right click the project->build configurations->build all.]
Missing file or directory
[image]

What can I do for next?

jantje, 2015-07-18:
please provide console output.

kshok, 2015-07-19:
Hi.
Console output is here

09:57:20 **** Build of configuration Debug for project debugDemo ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"/opt/arduino-1.6.1/hardware/jantje/avr_local/cores/arduino" -I"/opt/arduino-1.6.1/hardware/jantje/avr_local/variants/standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"   -Wall
In file included from /opt/arduino-1.6.1/hardware/jantje/avr_local/cores/arduino/Arduino.h:17:0,
                 from ../.ino.cpp:6:
/opt/arduino-1.6.1/hardware/jantje/avr_local/variants/standard/pins_arduino.h:28:26: fatal error: avr/pgmspace.h: 그런 파일이나 디렉터리가 없습니다
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
make: *** [.ino.cpp.o] Error 1

09:57:21 Build Finished (took 913ms)

jantje, 2015-07-21:
With me it is only a warning.
Please make an empty pgmspace.h file.
Best regards
Jantje

jantje, 2015-07-21:
so you can debug now?

kshok, 2015-07-21:
yes

jantje, 2015-07-21:
:-)
I'll change the title to add pgmspace.h file

kshok, 2015-07-21:
see make: *** [debugDemo.elf] 오류 1 (<---error 1)

jantje, 2015-07-21:
seems your version of gcc has a problem with the -mmcu part of the command while mine has none
>Starting combiner
"g++" -Os  -Wl,--gc-sections -mmcu=atmega328p  -o "/home/jan/workspaces/arduino/Robot/batterijlader/debug/batterijlader.exe"    ./ChargerBrains.cpp.o ./batterijlader.cpp.o  ./Libraries/gps_Library/gps_Library.cpp.o  ./Libraries/Wire/Wire.cpp.o  ./Libraries/VoltMeter/VoltMeter.cpp.o  ./Libraries/SoftwareSerial/SoftwareSerial.cpp.o  ./Libraries/Servo/Servo.cpp.o  ./Libraries/SerialStringReader/SerialStringReader.cpp.o  ./Libraries/SerialDataInterface/FieldInfo.cpp.o ./Libraries/SerialDataInterface/SerialDataInterface.cpp.o  ./Libraries/SerialCommunicator/SerialCommunicator.cpp.o  ./Libraries/SerialBridgeCommunicator/SerialBridgeCommunicator.cpp.o  ./Libraries/JanServo/JanServo.cpp.o  ./Libraries/I2CLiquidCrystal/I2CLiquidCrystal.cpp.o  ./Libraries/DataTypes/DataTypes.cpp.o  ./Libraries/CurrentSensor/AmpMeter.cpp.o ./Libraries/CurrentSensor/LemCurrentSensor.cpp.o   arduino.ar   "/home/jan/workspaces/arduino/Robot/batterijlader/debug/arduino.ar" "-L/home/jan/workspaces/arduino/Robot/batterijlader/debug" -lm -lpthread
Finished building: batterijlader.elf

can you mod the platform.txt and change 
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.exe" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm -lpthread

to 
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags}  {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.exe" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm -lpthread

open the project properties and select apply after changing the files to progress the changes to eclipse.

kshok, 2015-07-22:
Hi
I had mod successfully the platform.txt 
--------------------------------
Building target: debugDemo
Printing size:
ls -lrt "/home/ace/workspace/debugDemo/Debug/debugDemo.exe"
-rwxrwxr-x 1 ace ace 39860  7월 22 10:35 /home/ace/workspace/debugDemo/Debug/debugDemo.exe
Finished building target: debugDemo
--------------------------------
but popup error window .
[image]
how can I?

jantje, 2015-07-22:
https://cloud.githubusercontent.com/assets/13166935/8800996/3a860036-2ff2-11e5-9c65-4c68826cbb95.png
It looks like the gdb (dsf) process launcher doesn't work on your system.
I refer to the following quote in the blogpost http://blog.baeyens.it/#post23:
>Here you will have to experiment a bit with the settings. On my system I have to use the "legacy create process launcher". I use the settings as shown below.

Try the other debuggers until you find something that works on your system.

Please report what works for you.

kshok, 2015-07-22:
Wow!!
I did successfully Debugging as below.
change  "legacy create process launcher". 
[image]

also Debugging 
[image]

I will try Debugging arduino with EclipseMars for win7 and MinGW.
Thank your information.
Mr.KIm

jantje, 2015-07-22:
cool :-)

jantje, 2015-08-06:
Fixed the -mmcu and the missing pgmspace.h file

closed 2015-08-06 by jantje