jantje/hardware #3: Debugging arduino with EclipseMars for win7

kshok, 2015-07-26:
hi.
I try debugging arduino with 
EclipseMars(64bit) for win7 64bit.
Java 8.5.1 (64bit)
Arduino 1.6.1 (32bit)
MinGW  32  4.8.1

I was follow http://blog.baeyens.it/#comments23  .
also, https://github.com/jantje/hardware/issues/2 .

meet error
Console output is below
----
18:27:27 **** Incremental Build of configuration Release for project debugDemo ****
make all 
Starting combiner
"g++" -Os  -Wl,--gc-sections   -o "D:/WorkspaceArduino/debugDemo/Release/debugDemo.exe"    ./.ino.cpp.o   arduino.ar   "D:/WorkspaceArduino/debugDemo/Release/arduino.ar" "-LD:/WorkspaceArduino/debugDemo/Release" -lm -lpthread
c:/mingw/bin/../lib/gcc/mingw32/4.8.1/../../../../mingw32/bin/ld.exe: cannot find -lpthread
collect2.exe: error: ld returned 1 exit status
make: *** [debugDemo.elf] Error 1

18:27:29 Build Finished (took 2s.256ms)
here capture
---
[image]
Best regards
Mr.KIm

jantje, 2015-07-26:
looks like -lpthread needs to be replaced with -mthread when using mingw

kshok, 2015-07-27:
hi
I did change -lpthread to -mthread in platform.txt.
see  Console output is below
---
18:08:45 **** Incremental Build of configuration Debug for project debugDemo ****
make all 
Building file: C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/Arduino.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\Arduino\hardware\arduino\jantje\avr_local\cores\arduino" -I"C:\Arduino\hardware\arduino\jantje\avr_local\variants\standard" -MMD -MP -MF"arduino/Arduino.cpp.d" -MT"arduino/Arduino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/Arduino.cpp"  -o  "arduino/Arduino.cpp.o"   -Wall
In file included from c:\mingw\include\objbase.h:4:0,
                 from c:\mingw\include\ole2.h:9,
                 from c:\mingw\include\windows.h:114,
                 from C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/Arduino.cpp:3:
c:\mingw\include\rpcndr.h:52:23: error: conflicting declaration 'typedef unsigned char boolean'
 typedef unsigned char boolean;
                       ^
In file included from C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/Arduino.cpp:1:0:
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/Arduino.h:19:14: error: 'boolean' has a previous declaration as 'typedef bool boolean'
 typedef bool boolean;
              ^
make: *** [arduino/Arduino.cpp.o] Error 1

18:08:45 Build Finished (took 828ms)

Best regards
Mr.KIm

jantje, 2015-07-27:
I explained this before https://github.com/jantje/hardware/issues/1#issuecomment-119542791
I also do not understand why you had a linking problem and now a compilation problem.
Please do some more effort on trying to fix things.

kshok, 2015-07-28:
hi
I was reinstall Win7 (64bit)
And tested by several methods
Did not change liblary source file .
I did change -mthread to -lpthread  in platform.txt.
also I meet error
22:13:06 **** Incremental Build of configuration Debug for project debugDemo ****
make all 
other Building file is success...........................................................

Building file: C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10601 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\Arduino\hardware\arduino\jantje\avr_local\cores\arduino" -I"C:\Arduino\hardware\arduino\jantje\avr_local\variants\standard" -MMD -MP -MF"arduino/WMath.cpp.d" -MT"arduino/WMath.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp"  -o  "arduino/WMath.cpp.o"   -Wall
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp: In function 'void randomSeed(unsigned int)':
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp:33:17: error: 'srandom' was not declared in this scope
     srandom(seed);
                 ^
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp: In function 'long int random(long int)':
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp:42:17: error: too few arguments to function 'long int random(long int)'
   return random() % howbig;
                 ^
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp:37:6: note: declared here
 long random(long howbig)
      ^
C:/Arduino/hardware/arduino/jantje/avr_local/cores/arduino/WMath.cpp:43:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make: *** [arduino/WMath.cpp.o] Error 1

22:13:10 Build Finished (took 3s.736ms)

Thank & Regards
 Mr.KIm

jantje, 2015-07-28:
http://stackoverflow.com/questions/13174619/error-srandom-was-not-declared-in-this-scope

kshok, 2015-07-29:
Thank,  jantje  .
Now, my environment was constructed successfully.
It is your constant guidance help
I did below work.
change C:\Arduino\hardware\arduino\jantje\avr_local\platform.txt
---
1.  ## Compile c files  
 add  -Dsrandom=srand -Drandom=rand
2. ## Compile c++ files  
 add  -Dsrandom=srand -Drandom=rand
3. ## Combine gc-sections, archives, and objects  
 remove -lpthread     <--- Does it fit correctly ?   

debugging snap shot
---
[image]
remove -lpthread     <--- Does it fit correctly ?

thank & regards
 Mr.KIm

jantje, 2015-07-29:
cool :-D
If I have some time and energy somewhere I'll make a new version that works in linux and windows.
Thanks

jantje, 2015-08-06:
fixed

closed 2015-08-06 by jantje