jantje/hardware #4: Support for Teensy 3.1

hemantsangwan, 2015-08-26:
Hi Kindly add support for Teensy if possible. :)

jantje, 2015-08-26:
Teensy is supported.
http://trippylighting.com/teensy-arduino-ect/arduino-eclipse-plugin/

jantje, 2015-08-26:
oeps sorry; we are in the hardware section of github and not the plugin. 

What kind of support do you think is needed for Teensy? I'm not seeing the need.

hemantsangwan, 2015-08-26:
For debugging.

jantje, 2015-08-26:
>For debugging. Idea of Debugging and Unit Testing of Arduino code is awesome. 

I agree but what is so special about Teensy that it needs "support"
Have you read these blog pots? They explain what this is all about.

http://iloapp.baeyens.it/blog/blog?Home&post=19
http://iloapp.baeyens.it/blog/blog?Home&post=20
http://iloapp.baeyens.it/blog/blog?Home&post=23

hemantsangwan, 2015-08-26:
Great. It helped a lot. I will try to write down hardware files for teensy like hardware/jante. Any guideline? Suggestion? Tips? I am a noob in hardware files. Thanks in advance.

jantje, 2015-08-26:
> Any guideline?

~~In nice friendly letters:~~ Don't panic. ;:)
It are low-tech environment variable setting files.

jantje, 2018-03-22:
Did you get anywhere?