jantje/hardware #5: Debugging and unit testing Arduino (Part 3) issues & comments

pep, 2015-09-24:
Hi Jantje,
I tried to add a comment to Debugging and unit testing Arduino (Part 3) on you blog but I'm not a registered user.
I have been following parts 1-3 and have had some issues with getting it to work. I've numbered my problems, so its easier to respond to each: 

1/ in your refactoring of the original BlinkWithoutDelay code you have lost the setup() routine of the pinMode(LED_BUILTIN, OUTPUT); and as a consequence the UNO led does not flash properly when downloaded to the board. 
2/ You added a EDIT Note about setting up the gcc compiler in your path capable of making local executables. 
3/ I loaded cygwin and set the path, but 
4/ the debug build has issues knowing where to get specific headers from and it seems to cause a clash between the arduino, jantje hardware and the cydwin header/libraries. 
5/ Or is MingW preferred? 
6/ Maybe some more detail about how to configure each build configuration would be helpful. 
7/ Also I assumed that the Jante hardware files needed to be installed in the private libarary/hardware path. 
8/ When will the next installment part 4 be available to show the unit testing with googletest? 
9/ I notice that eclipse-arduino ide have the unit testing plugin component install but Im keen to see some examples. 
10/ Maybe we could use this project as a test project to the IDE test team, once it works fully? 
I sure everything will work its just some steps aren't too obvious as to how to control the library usage?
Regards, Phil.
phil.pidgeon@gmail.com

pep, 2015-09-24:
to clarify the clash: when building the debug configuration, I am getting errors with a 'conflicting declaration of C function 'int rand()' int _EXFUN(rand, (_VOID)); which looks like is coming from the /usr/include/stdlib.h in the cygwin libraries. Phil.

jantje, 2015-09-24:
1) indeed.
5) That user used mingw and I tested it on windows with mingw so mingw works. Does that make it preferred  ... ? I don't think so .... It makes it "tested"
6) I'm not sure what you are looking for.
7) It is not important where the libraries are. As long as the plugin finds them.
8) Good question. I have to many things at hand right now.
9) You noticed well. It is needed for 8.
10) It could be part of the testing. But the local debugging doesn't use upload so it is only part of the functionality and I fear not many users are really interested.

The problem you have with rand() is caused by the following info in platform.txt -Dsrandom=srand -Drandom=rand
If you remove those in your local platform.txt and reapply your project arduino settings it will probably work.
The root cause is that windows uses srand and rand while linux uses srandom and random. As cygwin is a linux simulator on windows it does not need the "renaming" and as such the renaming is creating conflicts.

jantje, 2015-09-24:
Why do you say "I tried to add a comment to Debugging and unit testing Arduino (Part 3) on you blog but I'm not a registered user."
Normally anyone should be able to post a comment.

pep, 2015-09-24:
Jantje, I added my user name, entered the magic number and my comment text and just got a message in red or orange? saying something like "comment text not accepted" I tried a couple of times and gave up on it, and then added this issue, as the first comment on the Part3 suggested?  I'll go through your suggestions today, hopefully it will all come together. Regards, Phil.

jantje, 2015-09-24:
It is all brand new so this is bleeding edge. However it should just work (within boundaries).
So if something doesn't work it is best to shout. Don't expect google to be helpfull.

pep, 2015-09-25:
Jantje,
regarding my previous post above:
6) What I was looking for is how to select the tool-chain for the debug configuration so the right libraries are chosen. I guess what I was asking is:

1/ What actually controls where the project includes come from? and, 
2/ how and when are they updated in the Eclipse IDE when things are changed? 
The reason I ask is I have seen them stay the same as previous toolchain compiler/build settings and not be updated.

When I installed windows MingW and Cygwin I put MingW on the path first:
PATH = ....;C:\MINGW\bin;C:\cygwin64\bin; on the end.
For the debug configuration I selected project/properties/C\C++ Build/Tool Chain Editor/Current toolchain selectbox MingW GCC. 

So what builder should I use with MinGW GCC? the options I get are: "Arduino sketch builder", "CDT Internal builder", "Gnu Make Builder", and"org.eclipse.linuxtools.cdt.autotools.core.toolchain.builder". 
I guess behind this selection it chooses specific build options and paths?

Then what should I set the Current builder to? 
	If I leave it at "Arduino sketch Builder" and build All I get a missing include for<avr/pgmspace.h>
	If I set it to "CDT Internal builder" I get Building debugDemo errors
	If I set it to Gnu Make builder and build All, and run debug I get a missing include for <avr/pgmspace.h> as well.

Othertimes it won't find the <iostream> library from HardwareSerial.h, and other times it won't find srandom for WMath.cpp?

if I try using cygwin the same builder options are available.

I think maybe using the "Gnu Make Builder" would be right but maybe the "Arduino sketch builder" is needed?

Regards,
Phil.

pep, 2015-09-25:
Jantje, 
Is it possible in the WMath.cpp to use srand and rand in a way that conditionally switches for linux and windows to avoid the issues? Do you develop only on linux, if so what is your preferred environment that everything is tested on? maybe in the platform text that conversion can be made conditional in some way? I think a similar thing is happening with the "del" command used in org.eclipse.cdt.core if "CDT internal builder" is used with MingW.
regards,
Phil.

jantje, 2015-09-25:
I really think you should reread the blog post and start again sticking closely to it.
Why? There is no need to change the toolchain. Just select a board from "/hardware/jantje/avr_local/boards.txt" as if it is a arduino boards.
Because of the way I set it up ()the configuration "debug" is specified in the "debug configuration"  there is no need to change the configuration. The idea is that it is all hidden and easy.

About how you see the code: Have you turned "use active build configuration" on for "build configuration for the indexer" in windows->preferences->C/C++->indexer?

>Is there a way to see what the actual command lines that are generated in the compile and build process (like a verbose or something?) because I'm not so sure that the -Dsrandom=srand -Drandom=rand are effective in some cases? I looked in the debug makfile and .mk files but couldn't see any?

Can't you see that in your console view? If not I don't understand what you are asking for.
Note that in this setup you have multple console views (at least one for each build configuration) so look at the right one.

> I'm not so sure that the -Dsrandom=srand -Drandom=rand are effective in some cases?

I fully agree. It is a fix I did for mingw which might not be needed for cygwin.  It is probably better to fix this in the code based on defines.

>If I leave it at "Arduino sketch Builder" and build All I get a missing include for avr/pgmspace.h

Are you sure you have downloaded the latest version? #2 
Right now avr/pgmspace.h is an empty file and I added it to the repository in a fix.

pep, 2015-09-26:
I have followed the blog post part 3 process closely. But my questions about includes and compiler/build have arisen where what I see when I perform the step differ slightly from the screen shots in the post. Where you suggest that things are happening "hidden and easy" may not be happening exactly as you expect. So I decided to do the whole process again and note where things are different and I noticed some things. I think that depending on the build configuration that is selected different things happen. For example, if one does a clean project only the build configuration selected is cleaned not the whole project. Also if a specific build is selected as active different things happen when a build all config is done. If a build selected config is done separately for each config the results are different too. I think that a very specific sequence of steps needs to be followed to get your required outcome and if the wrong build configuration is selected at the wrong time the results vary widely (missing includes iostream.h/pgmspace.h. for example the debug configuration created by the debug build configurations is different depending what configuration is selected with a wrong directory path and a missing .exe sometimes. All this is very confusing for a beginner not knowing what files in what configurations should have been created when? I'm going to try to follow step by step and make some specific observations of what isn't the same at specific process steps. It actually enough to send someone crazy......I collect my observations and try to document it in my next post to be more specific.

pep, 2015-09-26:
If I use build all I get the errors previously mentioned. BUT, If I build the configurations separately and don't use build all, each one builds individually. In the post part3 where you perform the "debug as"->debug configurations then Select C/C++ application and new. when I do it sometimes there is already a debugDemo Debug C/C++ application already created, other times not,  but either way the path is Release\debugDemo.elf NOT Debug\debugDemo.elf because there is no .elf created in the Debug directory. Then when I press the debug button if it isn't greyed/out the debug session starts but ends with a Problem Occurred dialog 'Launching debugDemo Debug' has encountered a problem."Target request failed: Error creating process Release\debugDemo.elf, (error 193).

pep, 2015-09-26:
I'm sure I have the latest jantje hardware as it has a an empty pgmspace.h file in it. I didn't select any toolchain in the debug configuration, but it still should the arduino toolchain/builder independent of whatever it does actually use based on the hardware.....

jantje, 2015-09-26:
>If I use build all I get the errors previously mentioned. BUT, If I build the configurations separately and don't use build all, each one builds individually.

Question: Do you use ino .files? 

What you should compare are the console outputs. There you see all the info being used by make. Without those I can only read your conclusions, I can not see what you get as a result.

>other times not, but either way the path is Release\debugDemo.elf NOT Debug\debugDemo.elf because there is no .elf created in the Debug directory.

http://iloapp.baeyens.it/blog/blog?ShowFile&image=1435787909.png states .exe. The path should be debug\debugDemo.exe. There is no way I know of to debug a .elf file on a pc.
I know that these multi build configurations and multi debug configurations can be confusing. For sure in a multiproject workspace. To get it to work I advice to have only one project with a release and a debug build configuration and one debug configuration. This so you can understand how it should work. Know that clicking the debug icon starts the latest debug configuration of the currently selected project. 
If you go through the debug configuration dialogue it is easier to understand what you will be debugging.
I think you are fighting a lack of understanding of the eclipse and cdt environment for this issue. Not so much the arduino eclipse plugin.

>I'm sure I have the latest jantje hardware as it has a an empty pgmspace.h file in it.

If the compile doesn't find pgmspace.h file; the please provide the console output. Without that I can not help you out. If this one is missing it is the arduino eclipse plugin.

pep, 2015-09-26:
Jantje,

Yes I'm using .ino files.

------------
Here is some console output for If Release config is active and build all done

01:34:11 **** Build of configuration Debug for project debugDemo ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"    -Wall
In file included from ../.ino.cpp:6:0:
C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\cores\arduino/Arduino.h:28:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
make: *** [.ino.cpp.o] Error 1

01:34:12 Build Finished (took 547ms)If Release config is active and build all done

01:34:11 **** Build of configuration Debug for project debugDemo ****
make all 
Building file: ../.ino.cpp
Starting C++ compile
"g++" -c -g -O0 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\cores\arduino" -I"C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp"  -o  ".ino.cpp.o"    -Wall
In file included from ../.ino.cpp:6:0:
C:\Users\Phil\Arduino1.6.5-r5\hardware\arduino\avr\cores\arduino/Arduino.h:28:26: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
make: *** [.ino.cpp.o] Error 1

01:34:12 Build Finished (took 547ms)

----------------------
You are right! it should be .exe, but for some reason this demoDebug debug configuration is being created sometimes with everything 'nearly' the same as what is needed, except the directory and the extension are different, creating confusion on my part. I just tried it again and edited that field to have the debug directory and the .exe extension, its also automatically now using the legacy create process launcher by default, which it wasn't doing before? 

I am also getting a Error in the Error Log. Error: java.io.IOEXception: Cannot run program "cygpath": Launching failed in org.eclipse.cdt.core, but I can run the Debug As demoDebug application and it stops at the Arduino.cpp file at the inVariant(); line. So I think I am making some progress....It maybe a problem with cygwin instead of MingW?

I'm also getting a warning: /cygdrive/c/.......directories in jantje/avr_local and debugDemo directories No such file or directory showing up numerous times in the console , looks like another cygwin compatibility issue?

-----------------------------
while running the Debug As....

.gdbinit: No such file or directory.
Reading symbols from C:\Users\Phil\Desktop\eclipseArduino\workspace1\debugDemo\Debug\debugDemo.exe...done.

Warning: /cygdrive/c/Users/Phil/Arduino1.6.5-r5/hardware/arduino/avr/variants/standard: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Arduino1.6.5-r5/hardware/arduino/avr/cores/arduino: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/arduino: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Release/arduino/core/avr: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Release/arduino/core: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Release/arduino: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Release: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Debug/arduino/core/avr: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Debug/arduino/core: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Debug/arduino: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/Debug: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo/.settings: No such file or directory.
Warning: /cygdrive/c/Users/Phil/Desktop/eclipseArduino/workspace1/debugDemo: No such file or directory.
[New Thread 6228.0x24f0]

Regards, Phil.

jantje, 2015-09-26:
using .ino files will make things harder because the standard build command does not generate/update the .ino.cpp file.
I remember someone else had this issue but I can't find it back so I created a new one https://github.com/jantje/arduino-eclipse-plugin/issues/311
But the .ino.cpp file is there so this is not your problem

It is clear that on your system, for whatever reason the build of debug is not done properly.
It seems like the build config switch was not executed completely. Part of it is done 
The command line goes to g++
The -DLOCAL_RUN is defined
but the includes point to arduino avr and not to jantje avr.
It may be related to cygwin but it is hard to tell from here.

As the build fails the .exe is not there so the debug doesn't start.
However

> I am also getting a Error in the Error Log. Error: java.io.IOEXception: Cannot run program "cygpath": Launching failed in org.eclipse.cdt.core, but I can run the Debug As demoDebug application and [...]

You did seem to get it into debugging.
I fear it is a cygwin problem or a cygwin MingW conflict you are looking at.

pep, 2015-09-26:
Jantje, How does the Eclipse Arduino IDE choose between the MingW or cygwin and how can one tell which is being used. the toolchains aren't updated in the debug configuration to show what is used, that would be nice to remove all ambiguity, and stop people thinking they should change it, if its already clear what is being used). ATM I have both installed with MingW first of the the path followed by cygwin64 second (maybe the 64bit versions might also have something to do with it?) At least I am getting into the debug now, which I wasn't even doing before. Many thanks. Do you have a very basic unit test src file for this debugDemo that I could try out? Regards, Phil.

pep, 2015-09-27:
Jantje,
do you have any recommendations on how to get familar with Eclipse CDT, ie books or specific tutorials that are very good, seminal, condensed to help me get in better alignment with your understand of the tool? I've been interested, collected some already, but really needed a reason, and something specific to focus on till now.
Regards,
Phil.

jantje, 2015-09-27:
>How does the Eclipse Arduino IDE choose between the MingW or cygwin? Is it just what it finds first? 

It is what the os finds first. As you have cygwin installed I guess you can do which g++ to find out what is used.
I can not recommend any sources for learning eclipse or cdt apart from the official documentation http://www.eclipse.org/cdt/. 
The readme.md at the bottom of this page https://github.com/jantje/arduino-eclipse-plugin explains exactly how to get the plugin source code on your system and compile. 
For coding you need the "Eclipse IDE for Java Developers" and add the pde plugin. 

My agenda is crowded for the moment so I won't have time to explain the unit testing, sorry.

jantje, 2018-03-22:
I'm back alive :-)
I see I did do some changes in this area. How is your status on behalf of this issue?