jantje/hardware #6: V4 ESP8266 Hardware Debug Works for 'Huzzah Breakout' board, but not for 'Huzzah Feather' board

mrwgx3, 2017-05-25:
Per Janjte's request, the following issue was closed in 'Sloeber/arduino-eclipse-plugin' and recreated here (san's Janjte's comments):

V4 ESP8266 Hardware Debug Works for 'Huzzah Breakout' board, but not for 'Huzzah Feather' board, #752

See  https://github.com/Sloeber/arduino-eclipse-plugin/issues/752

## 1st Comment

### Basic Infos

#### Hardware

```
Hardware:     ESP12, Adafruit Huzzah Breakout and Feather Boards  
Core Version: 2.3.0
Computer:     Dell, Intel Core i7 CPU, 870 @ 2.93Ghz,  16.0 GB RAM
```

#### OS, IDE

``` 
OS:   Windows 10 Professional   
IDE:  Sloeber	4.0.1.201705071425
```

### Description

Per your [Youtube video on how to setup Sloeber for hardware debugging of the ESP8266][debug_video], I successfully got debugging to work for the Adafruit [Huzzah Breakout Board][huzzah_breakout]. The [Huzzah Feather Board][huzzah_feather], however, seems to have timing issues with both the uploading of the debug sketch and initiating a debugger dialog.

[debug_video]: https://www.youtube.com/watch?v=S_QTMNhaDwM
[huzzah_breakout]: https://www.adafruit.com/product/2471
[huzzah_feather]: https://www.adafruit.com/product/2821

### Discussion
To facilitate testing, (2) 'blink' example projects were created:

```
   Debug Config.               Release Config.                 Serial     Project Title
A) Debug Board with 1.65 SDK   Huzzah Feather  with 1.65 SDK   COM4       blink_wtdelay_gdbA_r1p65
B) Debug Board with 1.65 SDK   Huzzah Breakout with 1.65 SDK   COM6       blink_wtdelay_gdbB_r1p65
```

Initially, only project A existed; project B was created after project A's debug sketch wouldn't upload. Project B's debug sketch did upload, and invoking the Debug perspective resulted in a successful debug session. Project A's debug sketch was then uploaded using the admin. command prompt to run a different 'esptool', but switching to the Debug perspective only resulted in a partial load and the following timeout message:

```
Error in final launch sequence
Failed to execute MI command:
-target-select remote /./COM4
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout
```
Hence it was concluded that the Huzzah Feather Board likely has some software/hardware timing issues.

### Messages of Interest

Project B Compiler/Upload Messages, Successful Debug Session
```
'Create eeprom image'
"C:\SloeberBeR1\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.5/esptool.exe" -eo "C:\SloeberBeR1\/arduinoPlugin/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/bootloaders/eboot/eboot.elf" -bo "C:/Users/DELL/SloeberWork/Wksr1/blink_wtdelay_gdbB_r1p65/debug/blink_wtdelay_gdbB_r1p65.bin" -bm qio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "C:/Users/DELL/SloeberWork/Wksr1/blink_wtdelay_gdbB_r1p65/debug/blink_wtdelay_gdbB_r1p65.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
'Finished building: blink_wtdelay_gdbB_r1p65.hex'
' '
'Building target: blink_wtdelay_gdbB_r1p65'
'Printing size:'
"C:\SloeberBeR1\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-size" -A "C:/Users/DELL/SloeberWork/Wksr1/blink_wtdelay_gdbB_r1p65/debug/blink_wtdelay_gdbB_r1p65.elf"
C:/Users/DELL/SloeberWork/Wksr1/blink_wtdelay_gdbB_r1p65/debug/blink_wtdelay_gdbB_r1p65.elf  :
section                           size         addr
.data                             1496   1073643520
.rodata                           1716   1073645024
.bss                             42192   1073646744
[... 248 lines cut ...]

C:\Users\DELL>C:\SloeberBeR1\arduinoPlugin\packages\esp8266\tools\esptool\0.4.9\esptool.exe -cd nodemcu -cb 921600 -cp COM8 -ca 0x00000 -cf C:\Users\DELL\SloeberWork\Wksr1\blink_wtdelay_gdbA_r1p65\debug\blink_wtdelay_gdbA_r1p65.bin
Uploading 216096 bytes from C:\Users\DELL\SloeberWork\Wksr1\blink_wtdelay_gdbA_r1p65\debug\blink_wtdelay_gdbA_r1p65.bin to flash at 0x00000000
................................................................................ [ 37% ]
................................................................................ [ 75% ]
....................................................                             [ 100% ]

Huzzah Breakout Board, stand-alone, expect this to work
C:\Users\DELL>C:\WINDOWS\system32>C:\SloeberBeR1\arduinoPlugin\packages\esp8266\tools\esptool\0.4.9\esptool.exe -cd nodemcu -cb 115200 -cp COM6 -ca 0x00000 -cf C:\Users\DELL\SloeberWork\Wksr1\blink_wtdelay_gdbA_r1p65\debug\blink_wtdelay_gdbA_r1p65.bin
Uploading 216096 bytes from C:\Users\DELL\SloeberWork\Wksr1\blink_wtdelay_gdbA_r1p65\debug\blink_wtdelay_gdbA_r1p65.bin to flash at 0x00000000
................................................................................ [ 37% ]
................................................................................ [ 75% ]
....................................................                             [ 100% ]

================================
```
This last test would indicate that the 'newer' esptool doesn't have an intrisic fault; it just needs different args.

Given this additional testing, its likely that
```
a) Preventing the 'newer' esptool from being invoked when the 'older' esptool is needed, and
b) Using a modified set of arguments when the 'newer' esptool is needed
```

would correct the problems seen the Huzzah Feather board.

jantje, 2017-05-25:
I propose to give it a try by using the old esp tool for all boards.
That is "how it should be" anyway.
Can you change in the platform.txt the line
`tools.esptool.path={runtime.tools.esptool-0.4.9.path}`
to
`tools.esptool.path={packages}/esp8266/tools/esptool/0.4.5 `
And test with both boards you have?

mrwgx3, 2017-05-25:
Made the requested change to 'platform.txt' and didn't see any change in 'esptool' arguments:
```
LaunchingC:\SloeberBeR1\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe -cd ck -cb 921600 -cp COM4 -ca 0x00000 -cf C:\Users\DELL\SloeberWork\Wksr1\blink_wtdelay_gdbA_r1p65/debug/blink_wtdelay_gdbA_r1p65.bin 
Output:
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
/arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe finished
upload done
```
To be sure we're both on the same page regarding file location and syntax...

 I changed file  'C:\SloeberHrdw\jantje\ESP8266\platform.txt', where 'C:\SloeberHrdw' is where I unzipped your hardware install.

Here are the changes I've made so far to 'platform.txt'. 
```
# ------------------------------

tools.esptool.cmd=esptool
tools.esptool.cmd.windows=esptool.exe

# Attempts to solve Huzzah Feather upload problem

#   Jantje's suggestion
tools.esptool.path={packages}/esp8266/tools/esptool/0.4.5

#   My attempt(s)
# tools.esptool.path={runtime.tools.esptool.path} 

# Original, comment out when testing alternate
# tools.esptool.path={runtime.tools.esptool-0.4.9.path}

tools.esptool.network_cmd=python
tools.esptool.network_cmd.windows=python.exe
```
Before doing a test, I close and clean all projects, exit Sloeber, then restart it.

I too did a correction attempt on this file by doing a file search on '0.4.9' to locate the file, but got the same result. Since it didn't work, I assumed I  entered an incorrect path specification and got a default invocation for 'esptool'.  If there is such a check, could it be failing due to a corrupt string? Erroneous conditional compile? Just tossing out ideas...

jantje, 2017-05-25:
have you done project properties->arduino->apply->ok?

mrwgx3, 2017-05-25:
```
 jantje commented 3 hours ago

 have you done project properties->arduino->apply->ok?
```
Didn't know that I needed to. Seems that I have to do this everytime I switch to a different project within the same workspace. Any way to do this automatically upon startup and/or change-of-sketch?

Progress has been made; I now see the expected 'esptool' version (old) and arguments, and the upload now occurs for the Huzzah Feather board. I still don't get a proper debug launch:
```
Error in final launch sequence
Failed to execute MI command:
-target-select remote /./COM4
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout
```

Note also that the blink LED turns on upon debugger launch, and turns off when the 'Problem Occured' message appears.

The Huzzah breakout board still works fine, good upload and debugger launch.

jantje, 2017-05-25:
You only need to do project properties->arduino->apply->ok when you changed one of the underlying files. Note that V4.1 is better at noticing change and doing this for you.
Did you see the comment of Paul van Veen?

> Use \.\COMX instead of /./ !﻿

mrwgx3, 2017-05-26:
No, I hadn't.
No difference between Use ' \.\COM4 instead of /./COM4', still getting 'Error in final launch sequence'.
Thanks for your help!

jantje, 2017-05-26:
We are way out of my comfort zone here. So I don't think I can help any further.
Someone more experienced in the whole gdb remote debugging will be more helpful.
If you find a solution please feed back so I can modify Sloeber or the documentation.

mrwgx3, 2017-05-26:
I've had a partial success, meaning that I've seen the debugger start correctly, but not consistently.

The whole problem to me seemed to be one of timing, i.e., things were just on the verge of working, but just not quite. I decided to do some baudrate tests, as this being the only timing parameter than be changed easily. To my surprise, I can get enter debugging mode if:
```
a) The upload baudrate is 115200 or below.
b) The debugger baudrate is also 115200 or below.
```
It's surprising that the upload baudrate is a variable here; I suspect it might have something to do with the reset function. If you have any information on adjusting serial timing in Eclipse/Sloeber, it would be useful.

Ultimately, I'll probably need a logic analyzer to resolve the issue, a task for a later day.

jantje, 2017-05-26:
You know far more about this than I do. Good :-) this way it can get fixed.
The debug launcher is from gnuarm https://github.com/gnuarmeclipse the gnuarm documentation http://gnuarmeclipse.github.io/ or  @ilg-ul can help you with " information on adjusting serial timing in Eclipse/Sloeber,"

jantje, 2018-03-22:
Is this still a issue? Did you fix it? Can you contribute the fix to the repository?

mrwgx3, 2018-03-22:
Hi @jantje 

Yes, I believe its still an issue,  but I haven't had the tools or time to look into it properly.  Given my  chedule, I'd mark this issue as 'needs to belooked into' and give it a low priority, or even close it.
Best regards,
mrwgx3

jantje, 2018-03-22:
I'm seriously out of my depth here so I propose to close it and we can always reopen.

closed 2018-03-22 by jantje

mrwgx3, 2018-03-22:
`
I'm seriously out of my depth here so I propose to close it and we can always reopen.
`
I concur, close the issue.