jantje/hardware #11: debug error on V4.2

lewiss66, 2018-04-10:
Attempting to performe a debug of the STM32F103C8 board for instance and after watching your video about "Debug your arduino code with the zero and Sloeber" I tried with this board and after setting all up I got an error saying:
``Error while launching command: $(eclipse_home)\arduinoPlugin\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-gdb.exe --version``

The setup path for the debug config are:
->config option:
-s ${openocd_path}/../share\openocd\scripts -f  \target\stm32f1x.cfg
->Executable:
$(eclipse_home)\arduinoPlugin\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\bin\arm-none-eabi-gdb.exe
Path for the OpenOCD folder:
C:\Embeded\sloeber\arduinoPlugin\packages\arduino\tools\openocd\0.9.0-arduino6-static\bin
Any Idea?
Thanks

lewiss66, 2018-04-10:
For info I started a thread about sloeber in the STM32 community :)
https://community.st.com/thread/48944-c-ide

jantje, 2018-04-10:
Hi thanks for your interest in sloeber and Jantjes hardware. 
Unfortunately at this point in time Andrew niel when he states

> There is no debug support for STM32.

That being said...it should be easy to set it up when you know how to do hardware debugging with STM32 (which I do not know).

Note you can still build locally to debug your business logic.
Also note there is a plugin called "eclipse C++ ide for arduino" which is a different beast.

lewiss66, 2018-04-10:
I still think sloeber is the way to go for stm32 since It's a C++ oriented IDE that makes programs much easier to implement. 

> Note you can still build locally to debug your business logic.
Do you have any link or tutorial on how to build locally? I'll be interesting in that.

jantje, 2018-04-10:
> sloeber is more convenient and active than "eclipse C++ ide for arduino".

I'm not going to disagree on these points ;-)

>Do you have any link or tutorial on how to build locally?
Here is the regression testing video. There must be one about local debug which is about the same thing https://www.youtube.com/edit?o=U&video_id=y9hlAUrtdgI

If you know someone who knows how to hardware debug stm32 and wants to put some time in it, I'll be happy to help.

lewiss66, 2018-04-11:
Well, I bet some are watching...but ok I'll ask.
Jantje your youtube link is not working, at list for me.

lewiss66, 2018-04-11:
Does your video showing "Debug your arduino code on your local machine" works for sloeber version 4.2 as well? I seem to have trouble at the debug conf step. degub is still blurred...

jantje, 2018-04-11:
Seems I didn't proof read well enough in my last posts. I did some edits.
Normally things should work in V4.2 but the nightly is always easier as I run the nightly and as such I can check quickly.
The fact that you can not click debug is because there is something wrong in the input data. If you look at the image you provide (that is really appreciated) you see at the top a red thingy with "program does not exist".
As you can see you have a test.bin and not a test.exe. I guess you changed release to be debug and test to be release (I know from experience)

lewiss66, 2018-04-11:
To prepare the debug as per your video I setup in the project/properties a new configuration as locadebug Active then once build performed I got in the localdebug directory a file test.bin. Earlier when setup was for release Active therewas a test.bin as well in the release directory. I don't see any test.exe!
I'm confused.
[image]

jantje, 2018-04-11:
Things go really fast inn this area. You can install the local debug stuff with a json file like all boards.
The nightly comes with the json url included.
Install the jantje local board (no other installs needed in windows) and make sure your debug config uses this board and has the debug option set (not the unit test)