jantje/hardware #14: SoftwareSerial does not compile with

jml3on, 2018-08-11:
setup: I have the blink project, with a Debug configuration. It compiles fine in Debug and I cna run unit tests.

When I add the #include "SoftwareSerial.h" in blink.ino, Debug compilation breaks.
```
08:41:47 **** Incremental Build of configuration Debug for project blink ****
"C:\\dev\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\dev\sloeber\arduinoPlugin\packages\Jantje\tools\MinGW\5.3.0/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_PC   -DLOCAL_RUN   -I"C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\cores\arduino" -I"C:\dev\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.21\variants\standard" -I"C:\dev\sloeber\arduinoPlugin\libraries\gtest\1.0.1" -I"C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\libraries\SoftwareSerial" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"   -Dsrandom=srand -Drandom=rand
'Finished building: ..\sloeber.ino.cpp'
' '
'Building file: C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\libraries\SoftwareSerial\SoftwareSerial.cpp'
'Starting C++ compile'
"C:\dev\sloeber\arduinoPlugin\packages\Jantje\tools\MinGW\5.3.0/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_PC   -DLOCAL_RUN   -I"C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\cores\arduino" -I"C:\dev\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.21\variants\standard" -I"C:\dev\sloeber\arduinoPlugin\libraries\gtest\1.0.1" -I"C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\libraries\SoftwareSerial" -MMD -MP -MF"libraries\SoftwareSerial\SoftwareSerial.cpp.d" -MT"libraries\SoftwareSerial\SoftwareSerial.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\libraries\SoftwareSerial\SoftwareSerial.cpp"  -o  "libraries\SoftwareSerial\SoftwareSerial.cpp.o"   -Dsrandom=srand -Drandom=rand
C:\dev\sloeber\arduinoPlugin\packages\Jantje\hardware\pc\1.0.2\libraries\SoftwareSerial\SoftwareSerial.cpp:41:27: fatal error: avr/interrupt.h: No such file or directory
compilation terminated.
libraries\SoftwareSerial\subdir.mk:18: recipe for target 'libraries\SoftwareSerial\SoftwareSerial.cpp.o' failed
make: *** [libraries\SoftwareSerial\SoftwareSerial.cpp.o] Error 1

08:41:47 Build Failed. 3 errors, 0 warnings. (took 812ms)
```

config & workspace shown below:

[image]

jantje, 2018-08-11:
This is absolutely 100% as I expected.
I don't use software serial so there is no "dummy" software serial.
You can either #ifdef the software serial include (and all consequences) or make a dummy software serial and make a pull request :-)

jml3on, 2018-08-11:
I started playing with arduino about 2 weeks ago, so, while I'd like to contribute, I'm gonna hold off on the dummy library idea. I don't think you want me to send you pull requests at this point :-)

For now, it seems that restricting this code to the Release config might do, as I don't need it for debug nor tests.
```
#if !defined(__IN_ECLIPSE__) && !defined(LOCAL_RUN) && !defined(UNITTEST)
```
seems to be doing the trick, keeping both the compiler and the indexer happy.

Out of curiosity, why is the library included under jantje/hardware if it doesn't work? Why not remove it?

Thanks.

jantje, 2018-08-11:
>Out of curiosity, why is the library included under jantje/hardware if it doesn't work? Why not remove it?

good question. I was not even aware it was there. Probably unfinished business.

closed 2018-08-11 by jml3on

jantje, 2018-08-14:
Seems like a good approach as it forces you to separate the business logic from the device actions