jantje/hardware #15: String((unsigned int)1) crashes with pc/1.0.2

jml3on, 2018-08-14:
Using the following code under a blink project in local debug mode.
```
void setup() {}

void loop() {
  String t((unsigned int)1);
}

```

That program crashes when run in local mode.
It works fine when removing the `unsigned` qualifier.

Under debugger, it appears that calling from String(unsigned int)
```
char *utoa (unsigned char __val, char *__s, unsigned char __radix)
{
	return utoa ((unsigned int) __val, __s, (int) __radix);
}
```
loops for ever.

This is because `utoa` is not defined in stdlib.h, like itoa` is:
```
_CRTIMP __cdecl __MINGW_NOTHROW  char* itoa (int, char*, int);
```

not sure what a proper fix is, as I'm not sure if this is a mingw or jantje issue.

jantje, 2018-08-14:
> not sure what a proper fix is, as I'm not sure if this is a mingw or jantje issue.

I don't know and I wont have time to look at it before half september.

jml3on, 2018-08-15:
this older issue seems relevant: https://github.com/esp8266/Arduino/issues/110