jantje/hardware #17: Error when build .exe file  (debug) mode

robopassio, 2018-08-30:
I try to follow the video to debug my code on Arduino UNO.
With release mode, I can get a successful compilation as well as flash the code to the target board (default config which is: ../arduino/hardware/avr/1.6.21).
But when I create the debug_confg which used the avr_local (.../hardware/jantje/avr_local), immediately errors occur.
The console as follows:

```
23:50:21 **** Incremental Build of configuration LocalDebug1 for project DebugTest ****
"L:\\preProcess\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"G:\eclipseLib\hardware\jantje\avr_local\cores\arduino" -I"G:\eclipseLib\hardware\jantje\avr_local\variants\standard" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"   -Dsrandom=srand -Drandom=rand
The system cannot find the path specified.
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

23:50:22 Build Failed. 2 errors, 0 warnings. (took 675ms)
```
I have taken a long time for this but still cannot solve.
Could you please help to check and give me a solution to this problem.

jantje, 2018-08-30:
I guess you didn't install the jantje package using the boards manager.

robopassio, 2018-08-30:
Yes, I think so, because I watch your tutorial: https://www.youtube.com/watch?v=ZN-rLSyhBN8
There is no step of installing jantje package using the board manager.
are there any further steps, please help me to make it clear.
Thank you!

jantje, 2018-08-31:
See the comments on the video. The video is from before I added the jsonf file https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json