jantje/hardware #18: fatal error: c_types.h: No such file or directory

Crazor, 2018-11-21:
I'm trying to follow your [video](https://www.youtube.com/watch?v=WFzVzgEA8Fo) on debugging esp8266 code. When I create and activate the debug configuration and switch the toolchain to yours, I get the following error on building:

```
08:26:39 **** Build of configuration Debug for project Test ****
make all 
Building file: ../Test.cpp
Starting C++ compile
"/Users/crazor/Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/crazor/Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-I/Users/crazor/Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-I/Users/crazor/Documents/sloeber-workspace/Test/Debug/core" -c -Wall -Wextra -g -ggdb -Og -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DDEBUG -DDEBUG_ESP_PORT=Serial -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP12"  -DESP8266   -I"/Users/crazor/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266" -I"/Users/crazor/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/variants/nodemcu" -MMD -MP -MF"Test.cpp.d" -MT"Test.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../Test.cpp"  -o  "Test.cpp.o"
In file included from /Users/crazor/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:38:0,
                 from ../Test.h:9,
                 from ../Test.cpp:1:
/Users/crazor/Applications/sloeber.app/Contents/Eclipse/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/esp8266_peri.h:24:21: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
                     ^
compilation terminated.
make: *** [Test.cpp.o] Error 1

08:26:39 Build Failed. 2 errors, 0 warnings. (took 184ms)
```

I'm trying this with a newly created project and I've started over several times, but I can't get past this. Any help would be appreciated!

adamsishere1, 2022-10-15:
Having the same problem, and trying to fix it but I am confused on what you said "downloading the master branch of this project and dropping it into sloeber's packages folder"

jantje, 2022-10-15:
You no longer need this esp debug board with the new sloeber
I should remove it.
The newest sloeber allows you to change the compile instructions in project properties->sloeber so you can easily introduce the debug options without having to go to a different board.