jantje/hardware #19: Unable to debug with ESP8266 core 2.4.2

Menion2k, 2018-12-19:
Hello
I am struggling in making the gdb debug working on my NodeMCU ESP-12E, using the test blink.ino, in Sloeber standalone.
I use the jantje GIT master (that works with core 2.4.2)
I have configured the target in order to add -ggdb -Og option
I built the sketch and upload it on the target, then tried to connect with a GDB debug target with the \.\com4 serial connection. Of course I set the host gdb to the jantje delivere one and set gdbcmds as well.
The debug target is set to stop on setup()
I try to attach to the target but after 10 seconds more or less I get this error:

```
Error in final launch sequence:

Failed to execute MI command:
-target-select remote .\\.\\com4
Error message from debugger back end:
Bogus trace status reply from target: timeout
Failed to execute MI command:
-target-select remote .\\.\\com4
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout
```

It seems that the com4 is correctly opened but there is no answer from the target.
I have followed the Sloeber video on youtube, so I really don't know what can be the problem
Thanks for the support

Menion2k, 2018-12-19:
Hello
Actually I have understood the problem, see https://github.com/jantje/hardware/issues/6
The problem comes when you flash with a baudrate higher thatn 115200, doing so you need to reset the board before being able to connect to the target.
This is somehow strange, since the serial port should be forcibly reconfigured in 115200 with this code:

```
#ifdef DEBUG
	uart_div_modify(0, UART_CLK_FREQ / 115200);
	Serial.begin(115200);
	gdbstub_init();
#endif
```

Bye

jantje, 2018-12-19:
I actually thinking about abandoning this repository for esp8266.

The Arduino flavour of esp8266 now comes with a gdbstub library and a debugger now.
Therefore adding -ggdb -Og to the command line is (as far as I can see) now the only added value for the jantjes esp8266 hardware repository.

My major concern is that stopping the "support" will be explained as "it doesn't work" :-(

Menion2k, 2018-12-19:
Well
You can write a clear statement and PR a demo sketch in esp8266 arduino BSP as example for debugging with GDB in Sloeber

jantje, 2018-12-19:
That coud be a option. Unfortunately there is more urgent work on my stack right now :-(